mist-devel/mist-binaries #92: minimig_mist_rtg_201006.rbf, Kick1.2 problem ?

micv2, 2020-10-09:
Hello, after the latest minimig_mist_rtg_201006.rbf update, I can no longer start with Kick1.2 (yellow screen). I used Kick1.2 for very old scene demos ADFs. Kick1.3 works, causes image errors in some old demos .. Best regards, MiC

robinsonb5, 2020-10-10:
The kickstart 1.2 problem has already been identified and fixed - so the next binary release will solve that.

Can you let us know which demos in particular are displaying incorrectly with kick1.3, and what's the last binary release that displayed them correctly?

micv2, 2020-10-10:
Hi, This intro doesn't work with kick 1.3. it is a known problem. With kick 1.2 you can enjoy it.

http://www.pouet.net/prod.php?which=66562

robinsonb5, 2020-10-10:
Ah, thank you. (I misunderstood.  I thought that as well as reporting the Kick1.2 problem, you were also reporting something that used to work with Kick1.3 and doesn't any more.)

Kick1.2 will work again in the next binary release.

closed 2020-10-10 by robinsonb5

micv2, 2020-10-10:
Many Thanks :)

Edited the first link .. "World of Wonders" is correct.

gyurco, 2020-10-10:
If you set up some slow RAM, it'll work (workaround until the next release).

micv2, 2020-10-10:
Thanks :)

 
 

Gesendet: Samstag, 10. Oktober 2020 um 16:06 Uhr