mist-devel/mist-binaries #93: C64 Neos mouse & REU

Julitium, 2020-10-11:
Hi.

I would like to know if it is possible to implement the Neos mouse and others (cbm we already have it) in the C64 core. Also if possible a REU Ram expansion. We would have functionality FC3, Geos, etc.

Thank you very much.

Julitium, 2020-12-13:
Hi.
Here is implementation data 8726 (MMU) For REU. 

[CSG8726R1-TechnicalReference-1.0.pdf](https://github.com/mist-devel/mist-binaries/files/5684202/CSG8726R1-TechnicalReference-1.0.pdf)

gyurco, 2020-12-13:
It's a good reference, but I don't think I have to be the only one who works on this.

Julitium, 2020-12-13:
That's true, but you are a genius. Maybe it was not be very complex.
See if anyone is encouraged!!! . 
The SID would also be time to improve it, the updates of the VIC2 made by you put the mark very High. Thanks. 

Regards.

Julitium, 2022-01-01:
Hello.
@gyurco Thank you very much for the Christmas present.
Infinitely grateful for a core that already has all the improvements.
Merry Christmas. Happy New Year.
All the best.

closed 2022-01-01 by Julitium