mist-devel/mist-binaries #94: New core Minimig-AGA Gltch & Freezes

Julitium, 2020-10-18:
Hi.

The new Minimig-AGA core produces graphical glitch (Sysinfo) and crashes in games that need decrunch.
It looks like a Chip memory issue.
I attach an example photo.

Thank you.
[image]

gyurco, 2020-10-18:
Is this happen with turbo chip/kick ON, too?

Julitium, 2020-10-18:
Hi.
Yes. Too

Regards.

gyurco, 2020-10-18:
And with turbo OFF/Fast RAM OFF/only Chip+SlowRAM?

Julitium, 2020-10-18:
Turbo OFF / Fast OFF / 2Mb Chip/ 1. 5  Mb Slow.... Works fine. But poor result.

Regards.

gyurco, 2020-10-18:
Please try this.
[minimig_mist_issu94.zip](https://github.com/mist-devel/mist-binaries/files/5397950/minimig_mist_issu94.zip)

Julitium, 2020-10-18:
Hi.
Same issue. 
[image]

gyurco, 2020-10-18:
And with Turbo ON(Both), but Fast OFF (Only Chip+Slow)?

Julitium, 2020-10-18:
Turbo On (Both), Chip + Slow on, Fast OFF.
Works Fine. Same poor results. 
[image]

Julitium, 2020-10-18:
Turbo Off , Chip + Slow on, Fast OFF.
Works Fine. Same poor results.
[image]
Turbo On Both , Chip + Slow on, Fast OFF.
Works Fine. Same poor results. 
[image]

gyurco, 2020-10-18:
Maybe this one?
[minimig_mist_issu94_2.zip](https://github.com/mist-devel/mist-binaries/files/5398194/minimig_mist_issu94_2.zip)

Julitium, 2020-10-18:
Same results. No support FastRAM On.

Julitium, 2020-10-18:
Attached screenshot of the previous version of core in case it would help.
[image]

Julitium, 2020-10-18:
[image]
Conditions to Work: changing to previous core and then new core.

Julitium, 2020-10-18:
Previous version also Fortuitous.

gyurco, 2020-10-19:
Sorry, I have no idea, and cannot reproduce. If you can upload a hdf image (a not too large one), I can test with that.

gyurco, 2020-10-19:
Another idea: is this happens with both NTSC and PAL clocks? (force it in mist.ini clock_freq 1/2)

Julitium, 2020-10-19:
Hi.
I give you a HDD Image to Testing.

[HDD test.zip](https://github.com/mist-devel/mist-binaries/files/5404240/HDD.test.zip)

I will test to change clk_force.

Thanks.

gyurco, 2020-10-19:
Sorry, but it's unusable.
"Please insert a volume containing LIBS/icon.library in any drive"

Julitium, 2020-10-19:
[HDD_Test2.zip](https://github.com/mist-devel/mist-binaries/files/5404455/HDD_Test2.zip)
try this.

Kick 3.1

Julitium, 2020-10-19:
Clock Freq. 0/1/2 same results.

gyurco, 2020-10-19:
The test image works flawlessly - still no idea.

Julitium, 2020-10-19:
Uppps.... 
I have reason to think that it is precisely the HDF image that is wrong.
Maybe they mixed several versions.
Tested over others images and Works fine. 

Still Testing. 

Thanks Gyurco.

Julitium, 2020-10-19:
Confirmed ...
It's image failure.

Sysinfo 4.0 over Personal Image Works great.

Excuse my clumsiness, again you save me. I owe you a few beers.

Thank you very much.

gyurco, 2020-10-19:
Ah, good.

closed 2020-10-19 by gyurco