mist-devel/mist-binaries #96: Minimig-AGA P96: "No modes found"

sardaukar, 2021-01-27:
I've installed Picasso96 and followed the readme on the `MinimigUtils.adf` disk, but when I reboot and go into P96 Settings, I get "No modes found" and the UI won't let me add one.

Any help appreciated, thanks.

robinsonb5, 2021-01-27:
It sounds like it's not finding the Picasso96Settings file.
Go to the devs:monitors directory, click on the Minimig monitor icon, to to the "Icon" menu and select "Information" - what is the SETTINGSFILE tooltype set to?  It should be sys:devs/Picaso96Settings - or the path of your Picasso96Settings file if you put it somewhere different.

sorgelig, 2021-01-27:
Please move to MiSTer forum: https://misterfpga.org/viewforum.php?f=4
Probably there is already answer for your problem exists.

Keep github for development and bugs reports only.

robinsonb5, 2021-01-27:
Good advice, wrong platform.  For MiST go to https://www.atari-forum.com/viewforum.php?f=115

sorgelig, 2021-01-27:
oops.. i'm really sorry.. I have tons of notifications, so messed the platform..

sardaukar, 2021-01-28:
@robinsonb5 that is the path the monitor file had in the icon information. I'll try again and check the forum if I have any other problems. Thanks

closed 2021-01-28 by sardaukar