mist-devel/mist-binaries #98: mistery viking issue

Televicious, 2021-02-03:
new mistery core turns text yellow and locks up at boot with viking driver loaded and viking mode turned on.

gyurco, 2021-02-03:
With the FX68K CPU (68000 mode)?

Televicious, 2021-02-03:
Yea, set as MegaSTE chipset. Mistery version doesn't have the 020 option does it? Thought it was 68000 only. The old pre fxcast core is in 020 mode.

gyurco, 2021-02-03:
https://github.com/mist-devel/mist-binaries/blob/master/cores/Changes.md

MiSTery-210130.rbf

    Blitter force dest read fix regarding to endmasks
    Add selectable 68020 CPU (TG68K)

Maybe your setting is 68020, just you didn't update the firmware, and it's hidden.

Televicious, 2021-02-03:
Oh i did not check the 020 option. I merely updated firmware and when I ran into the issue compared to previous builds. Will check again.

gyurco, 2021-07-15:
Should be fixed in the latest core.

closed 2021-07-15 by gyurco