mist-devel/mist-binaries #99: C64 Tape Bugs.

Julitium, 2021-02-28:
Hi.
The C64 Core needs an update to the tape routines.
I have detected several pieces of software that do not work, they use the Ocean New Tape 4 loader (the ones listed in the link).

http://www.luigidifraia.com/c64/docs/tapeloaders.html#ocean4

In tests I have done on a C64 with Tapuino, there are some bass pilot tones that do not appear on the Core C64.
I have also detected some US Gold (Line of Fire) game, but I don't know what loader is usted. But same case. 
All the best. Thanks.

gyurco, 2021-03-02:
Try today's release, it loads Lethal Weapon for me.

Julitium, 2021-03-03:
Hi.
Everything is great. 100% surprised. @gyurco, you are undoubtedly a man of great talent.
Thanks for including the on / off switch in the progress bar (please remember to put it in the CPC core).
Thanks a lot.
All the best.

Julitium, 2021-03-16:
Hello.

Titles that previously failed now work fine. But I have checked others that now do not work (octoplex from mastertronic for example).
It seems to be when this jumps to the program code, apparently the load in the previous core is the same (but it works). 

Thanks a lot.
Greetings.

gyurco, 2021-03-16:
Try the attached core, it loads Octoplex. I don't know what other titles should I try.

[C64_mist.zip](https://github.com/mist-devel/mist-binaries/files/6151071/C64_mist.zip)

Julitium, 2021-03-16:
Hi.
All the tests I am doing are ok. It is at the release level.
I will continue testing.
Thank you very much @gyurco. 
All the best

Julitium, 2021-03-19:
Hello.
All main tests are Ok for me.
Thank you very much @gyurco. 
All the best.

closed 2021-03-20 by Julitium

gyurco, 2021-03-20:
Just changed a number :)
https://github.com/mist-devel/c64/commit/4325ee6b6872516de9c2e0e438994eb776bc3841