mist-devel/mist-binaries #102: HID Keyboard issue

AtalanteMIST, 2021-04-04:
Hello. I think there is some difference in HID between MIST and PC. I put my MIST in the Amiga 500 Case. I also used Amiga keyboard via Arduino Leonardo as a controller. It works, but not entirely. Both Shifts, ctrl, "Amiga" (on minimig core) doesn't work. I checked it on Minimig and Mistery cores. Everything seems to be ok if I connect it to the PC or RPI.

harbaum, 2021-04-05:
What firmware do you use on the Arduino? Does it implement boot mode? For keyboards MIST uses HID boot mode while PCs will use the full descriptors.

AtalanteMIST, 2021-04-05:
Thank you Harbaum for  being interested in my problem. 
By firmware you mean script I have uploaded? I checked two of them, from that sites:

[https://www.retro32.com/download/amiga-500-1000-2000-3000-4000-cd32-ctv-arduino-leonardo-pro-micro-keyboard-interface](url)
[https://forum.arduino.cc/index.php?action=dlattach;topic=139358.0;attach=276731](url)

Both behavies the same. Really don't know does it containes boot mode. How could I check that?

AtalanteMIST, 2021-04-05:
I tried to add mist.ini file. Inside was "keyrah_mode" option. As a parameter i tried  "0", "1", "18d80002", "23418037", "0x18d80002",  "0x23418037" unfortunately with no result. Maybe I'm doing something wrong?

gyurco, 2021-04-10:
Shift also doesn't work in any 8 bit cores, like C64 or the Spectrum?

AtalanteMIST, 2021-04-10:
I have checked also on Atari ST and Atari 800 core and it's the same.

robinsonb5, 2021-04-11:
Gyurco and I have been discussing this issue on atari-forum - https://www.atari-forum.com/viewtopic.php?f=115&t=40359
Could you try this update to mist-firmware, please?
[keyboardfix2.zip](https://github.com/mist-devel/mist-binaries/files/6291333/keyboardfix2.zip)

gyurco, 2021-07-15:
Should be fixed already.

closed 2021-07-15 by gyurco