mist-devel/mist-binaries #103: Feature request: more per core settings override sections

boomlinde, 2021-04-07:
I understand from the wiki that some cores allow settings overrides that apply to just those cores. Listed as supported are MINIMIG, ST, PACE, ARCHIE. Is it easy to extend this to more cores? My specific conundrum is that I have a Megadrive-style joypad with six face buttons (+ start and mode) and two extra shoulder buttons. I was thinking that I could comfortably use this for both the fpgagen and snes cores, but they would require different mappings.

gyurco, 2021-04-09:
It's possible to override any [mist] setting in a [corename] section. "Corename" is in the left side of the OSD.

boomlinde, 2021-04-10:
Thank you, that works just as I'd hoped! I appreciate the help.

closed 2021-04-10 by boomlinde