mist-devel/mist-binaries #105: SMS FM sound hiss a lot

PepAlacant, 2021-04-19:
The FM Sound of Sega Master System core hiss a lot. I

robinsonb5, 2021-04-19:
Can you give an example of a game where we can hear the hiss, please?

Is it something that happens as quiet sounds fade out?

jotego, 2021-04-20:
There's a good chance that the problem comes from the 1-bit DAC used.

PepAlacant, 2021-04-20:
There is a high evidence in fade sounds. It's very evident in games like Double Dragon and Altered Beast. 

https://youtu.be/2C0PHAD8U-Y
https://youtu.be/KNS4qcfp94k

robinsonb5, 2021-04-20:
> There's a good chance that the problem comes from the 1-bit DAC used.

Yes, that was my thought too - also the audio level seems to be rather low with this core.  Primitive DAC and quiet sounds - not a good combination!

> There is a high evidence in fade sounds. It's very evident in games like Double Dragon and Altered Beast.
> 
> in these videos is how the FM would sound correctly.

Thanks, that helps.

Can you try this core, and let me know if it's better, please?
[sms_mist_hybridsd_20210420.zip](https://github.com/mist-devel/mist-binaries/files/6341758/sms_mist_hybridsd_20210420.zip)

PepAlacant, 2021-04-20:
Yes, i'll test in a few mins. Thanks.

PepAlacant, 2021-04-20:
Now it sounds very good. It was about the general volume in the core ?

robinsonb5, 2021-04-20:
The core's low volume output is the reason it was sounding so bad - the simplistic 1-bit DAC used in a lot of cores is fine for simple full-volume sounds but performs poorly with quiet sounds.  Years ago I explored this effect on Minimig and came up with a hybrid PWM/Sigma-Delta DAC which helps a lot - and all I did for the core I posted here is swapped to the hybrid DAC.

But I never really understood *why* it helped as much as it does until very recently.  (In short, the simpler DACs are making assumptions about the characteristics of the FPGA's output pins which simply aren't true in practice.  The hybrid DAC is affected much less by this.)

@jotego you might find this interesting if you haven't already seen it:  http://retroramblings.net/?p=1578
I'm currently experimenting with a second-order hybrid variant which performs even better at low volumes.

The SMS core's output is still very quiet, though - I'll look at boosting it.

robinsonb5, 2021-04-20:
Here's a version with the volume boosted by 4x - which brings its audio output more in line with other cores.  Could you test it, please, and let me know if you hear any issues?

(I believe the mixed audio is correctly clamped to prevent it overflowing but I haven't found anything loud enough to test that yet - it might never happen, but if you find anything that sounds horribly distorted with this version please let me know.)

[sms_mist_hybridsd_volboost_20210420.zip](https://github.com/mist-devel/mist-binaries/files/6342588/sms_mist_hybridsd_volboost_20210420.zip)

Julitium, 2021-04-20:
Hi.
Try Wonder Boy 3 Dragons Trap.... Its good FM test.
Regards.

robinsonb5, 2021-04-23:
> Tested.... Great Sound!!

Thanks for testing - here's an updated version in sync with upstream.  (The previous version had some menu items commented out since I used my TC64 tree for the test build.)
[sms_mist_20210423_upstream_hybridsd_volboost.zip](https://github.com/mist-devel/mist-binaries/files/6364107/sms_mist_20210423_upstream_hybridsd_volboost.zip)

closed 2021-04-25 by gyurco