mist-devel/mist-binaries #106: FPGA64 SID Code.

Julitium, 2021-04-26:
Hi.
After hearing many rumors (and failed tests) I have seen that @Sorgelig has included new code to implement the SID. I would like to know if it is possible to extrapolate that code to our beloved MiST version that so brilliantly updates the Great @Gyurco.
Thank you very much for your opinion and reading this query (once it is cleared it will be deleted for "mailbox hygiene".

All the best

gyurco, 2021-04-27:
Why not? But I would prefer if someone else with more SID experience would do it. With the VIC-II it's rather obvious if the result is good or bad, with the SID, it's much more subjective.

Julitium, 2021-04-28:
Thank you very much for your answer.
Thanks again for your work on core 64, it's absolutely great

closed 2021-04-28 by Julitium