mist-devel/mist-binaries #107: PC Engine CD Issues

PepAlacant, 2021-05-26:
Flash Hiders don't boot ( Black Screen )
Popful Mail start game intro's is completely missing
Kabuki Ittoryoudan very initial initial intro and Ane-san start screen is almost invisible. Brightness and contrasts gets down.
The BIOS screen has a glitch in the "Press Run Button" letters

gyurco, 2021-05-26:
Flash Hider's bug was an omission of a BCD->binary conversion, fixed.
I cannot find a Popful Mail dump with single image.
The low brightness issues is when the game draws to the back porch area, maybe forcefully blanking this area is required.
The glitch in BIOS doesn't present in the MiSTer version? I don't see obvious issues with the VDC port.

PepAlacant, 2021-05-26:
The MiSTer version doesn't have that glitch. It's like a corruption when the letters appear and dissapear. They are like artifacts that appear here and there in the "push run button" words.
The Popful Mail game has issues in the Terraonion's Super SD System 3 too. Related to the intro that appears when you press start to play the game ( Corrupted Graphics in this case ) 

Popful Mail and Flash Hiders are optionally compatible with Arcade Card. The Arcade is not needed to play them, but it gives more RAM space for the games.
MiSTer works perfectly with all those games.

PepAlacant, 2021-05-26:
There is another issue. 

Gulliver Boy

The games uses a feature called Hu-Video. It requires accurate seek and reading CD. 

The intro of the game is a Full Motion Video that shows a lot of errors in MIST core. The intro of the game works fine in MiSTer.

This error can affect the Sherlock Holmes' games in PC-Engine CD too.

gyurco, 2021-05-26:
World Heroes 2 works for me with enabled Arcade Card.
Maybe it's because of the bugfixed GETDIRINFO in the firmware. Attached it if you want to try.
[fw-lba.zip](https://github.com/mist-devel/mist-binaries/files/6549301/fw-lba.zip)

PepAlacant, 2021-05-26:
I could make work World Heroes but, doing a nasty thing. Maybe it's my ISO, that it's not correct. When trying to load the PLAYER SELECT I reloaded the img and it worked. The real machine does a full stop of the CD at this point. I perceived small flashing line glitches in the hud, same as the BIOS " Push Run Button", but nothing really important.

Flash Hiders is working perfect now. The rest of the issues are present.

Thanks a lot for the Firmware. I'll test more, and if I find something I will post it here.

gyurco, 2021-05-27:
Adding delays before serving a sector helps more or less on video playback glitches, but it's hard to make it perfect.

PepAlacant, 2021-05-27:
Yes, the Hu-Video is maybe the most accuracy demanding on the CD-ROM reading. All games I could test are working. These glitches I told you appear, but I'm thinking that maybe my MIST model, as it is a MiSTICA16, would need a tweak in the build about the ram access, I will test more. 

Thanks again for the hard work, Gyurco !

gyurco, 2021-05-28:
The attached core/firmware fix most (all?) of these issues for me.
Try them if you want.
[tgfx16_mist_210528.zip](https://github.com/mist-devel/mist-binaries/files/6561966/tgfx16_mist_210528.zip)

PepAlacant, 2021-05-28:
Excellent! I'm testing right now and some of the issue are not anymore. Gilliverboy intro is shown. The only thing is he Popful Mail intro that doesn't appear, but all the glitches and things like that are completely gone.

gyurco, 2021-05-29:
Popful Mail intro is playing for me, although I don't understand a word in it...
Some puppets has stolen a treasure box.

PepAlacant, 2021-05-29:
Yes, it's the first intro. The puppets steal a treasure box. When you push run, it goes to the selection screen. Then when press beginning, another intro appears (that's the one that is missing ) only the first scene is shown and the rest is completely black.

PepAlacant, 2021-05-29:
Great work with hu-video fixing.

gyurco, 2021-05-29:
I see. This one plays that intro!
[tgfx16_mist_210529.zip](https://github.com/mist-devel/mist-binaries/files/6564886/tgfx16_mist_210529.zip)

PepAlacant, 2021-05-29:
Okk! I could test it and this is working excellent. There is a last thing I noticed. This I can't say exactly if it's a problem of my ISO or a problem in the core or the firmware. I noticed in two games.

Gradius II Gofer no Yabou
4 in 1 Super CD ( the bundle disk for turboduo in USA )

When the track is very far from the first track of the disk. It can be lost or not found. In the original hardware this takes long to go from the first track to the last. If the lens is in not very good condition, sometimes gets lost. 

These two games have this behaviour in the intro, in the case of Gradius II (track name: star flight) and when Bonk appear to present the 4 in 1 compilation it goes to a very far track that are the sound of applause.

I will test other isos for discard.

Thanks a lot for your work. This core is being fantastic.

gyurco, 2023-04-26:
I close this.
If other issues are found, please open a bugreport at https://github.com/mist-devel/TurboGrafx16_FPGA

closed 2023-04-26 by gyurco