mist-devel/mist-binaries #108: Thanks Gyurco.

Julitium, 2021-05-26:
Hello.

I would like to thank @gyurco for all the work he has done on all MiST cores especially on C64, Spectrum and Amstrad CPC.
Now it has surprised me again by including the CD support that I am sure will give new possibilities to existing cores (SegaCD, CD32, etc) and will open new horizons (PSX, 3DO, etc).
I would like to encourage the developer community to continue to release cores with such quality.

Thank you very much @gyurco (& develop community).

gyurco, 2021-05-27:
I'm glad you're excited but these more advanced cores are unlikely to happen on the original MiST.

Julitium, 2021-05-31:
Hi.
Perhaps someone will be encouraged to introduce the CD routines to core Megadrive (Mega CD) and Amiga (CD32). I think the power of MiST allows it.
Thank you again ;-) 

Greetings

gyurco, 2021-06-02:
The FPGA might be enough, but the enormous amount of RAM bandwidth required by the MegaCD hardware looks too much for that poor single SDRAM chip.

Julitium, 2021-06-02:
Hi.
I understand. The numbers were adjusted by the "grandmother's count" ...
if the Amiga core at full power sysinfo says it is equivalent to a 68020 @ 99Mhz, a 68000 @ 7Mhz and a 68000 @ 12 Mhz it would be possible, but I did not fall for the RAM bandwidth issue.
Ahhhggg :(
Thanks again for your work.
All the best

alessioscand, 2021-07-29:
I finally found a way to say THANK YOU to @gyurco for the Next186 core on Mist.
Thank you man, really, thank you so much.

gyurco, 2021-07-29:
@alessioscand You're welcome! However the hard work was not the porting, but the work of the original author. Unfortunately this work is not finished, and for me, this core is a bit of disappointment, as it's not on par in compatibility with most of the MiST cores.

alessioscand, 2021-07-30:
@gyurco you're right, but with some tweaks this core can really become a true gem, as it already runs some early PC titles titles hard-to-manage on real hardware. Please keep it going! Thank you so much!

closed 2022-01-17 by gyurco