mist-devel/mist-binaries #109: Retroflag SNES USB gamepad support

z00m128, 2021-06-14:
Hello, I recently bought cheap "[Retroflag SNES classic USB gamepad](https://rpishop.cz/herni-ovladace/2462-retro-snes-usb-gamepad-pro-raspberry-pi-6971727910061.html)" which sadly does not work at all with MiST. With the MiST core and USB debug setting I got this output:
[Retroflag-gamepad-MiST-log.txt](https://github.com/mist-devel/mist-binaries/files/6648031/Retroflag-gamepad-MiST-log.txt)

Then I connected it into my laptop and found out that the gamepad is detected as Microsoft Xbox360 Controller:
[Retroflag-gamepad-dmesg-log.txt](https://github.com/mist-devel/mist-binaries/files/6648055/Retroflag-gamepad-dmesg-log.txt)

Is there any chance to get this thing working with MiST?
Thank you.

gyurco, 2021-06-14:
These connected-disconnected events happens rapidly? Then probably a powering issue.

z00m128, 2021-06-14:
No, only when it is plugged in. Then it looks like it is stabilized. Now I realized, that the gamepad can be set to HID mode (press and hold Y and plug in). In the F12/Input devices/USB status menu of minimig core I can see that it is somehow mapped, because the DB9 Joys are remapped to Joy2&Joy3.

Also the _mapper.exe_ can see it afterwards and produces _joystick_remap=0583,2060,1,2,4,8,10,20,100,200,400,800,40,80_, which i added to _mist.ini_, but still no luck and gamepad is still not working.

The debug output of MiST core looks like this now:
[Retroflag-gamepad-HID-log.txt](https://github.com/mist-devel/mist-binaries/files/6648814/Retroflag-gamepad-HID-log.txt)

gyurco, 2021-06-14:
Then something must be seen on the Joystick 1 Test screen (in any generic core, like C64, fpgagen, etc...)

z00m128, 2021-06-14:
I don't know what I'm doing wrong but nothing is there. See the attached images.
[image]
[image]

gazsp, 2021-07-24:
Just to keep this alive - I have the same issue here.

gyurco, 2021-07-24:
Definitely an USB problem, just don't know where. It's hard to debug without an actual hardware.

Julitium, 2021-07-24:
Hi.
Perhaps it should be tried with the "Joystick Mapper" utility for PC. It's in main Git.
Maybe giving it the ID names, etc in Mist.ini Produces some improvement.

All the Best.

z00m128, 2021-07-24:
> Definitely an USB problem, just don't know where. It's hard to debug without an actual hardware.

If you want to investigate this, I can send the gamepad to you for free.

gazsp, 2021-07-24:
> Perhaps it should be tried with the "Joystick Mapper" utility for PC. It's in main Git.

Have done (as @z00m128 did). I see the same thing - Joy 1 with an empty name, and the tester shows nothing. I've attached my config file.

I've got a spare Retroflag Classic controller here - happy to post it to you for testing if you want to drop me an email (gary dot renes at gmail dot com).

[mist.ini.txt](https://github.com/mist-devel/mist-binaries/files/6872859/mist.ini.txt)

gazsp, 2021-07-29:
@gyurco Gamepad will be posted tomorrow. Good luck!

gyurco, 2021-07-29:
@gazsp Great, but if you need it back, then please don't send it.

gazsp, 2021-07-29:
No worries - it'll be in the post tomorrow.

gyurco, 2021-08-13:
Ok, the problem seems to be this:
HID: HID protocol is NONE
There's a hack for devices which identifies themselfes as "NONE" to be handled as Arduino-based keyboards, however seems it's not enough, as this gamepad also behaves similarly.

gyurco, 2021-08-30:
Looks like I have a success.
This gamepad is very strange, it has 3 personalities:
- an XBOX360 controller, which is not HID at all, it requires a totally different driver, which might be written for MiST in the future, but not in the near one
- "Nintendo Pro Controller" personality
- "RetroFlag Wired Controller"

The Nintendo one usually just initializes, then reset itself to the 3rd one, and that one works now. But only if it is the one which is booted. If it boots into the Nintendo one, then switches into Retroflag, it won't work. So it might need to press reset after turning on the MiST. Weird controller.

gazsp, 2021-08-31:
Nice work.

It is possible to make it switch modes by holding down a different button on the pad when powering up (think the info is on the box - I can't remember which buttons at the moment - X and Y maybe?). No idea if that helps or not...

gyurco, 2021-08-31:
Seems even you switch roles, the first powerup happens in Nintendo mode, then the gamepad resets itself, and goes into the Retroflag mode. In this case, it won't send the movement packets. However if you reset MiST, then the gamepad will start directly in Retroflag mode, and works. That's the best I could do so far, you have to reset MiST once after you turn it on.

gazsp, 2021-09-01:
Cool, I'll give it a shot :-)

gyurco, 2021-09-05:
You can try the attached firmware. Just press reset (left button) after the first power-up or gamepad connection (and switch to HID mode before - that's remembered by the controller, it has to be done once).
[firmware-usb-hat-joy-test.zip](https://github.com/mist-devel/mist-binaries/files/7112144/firmware-usb-hat-joy-test.zip)

mapfde, 2021-09-24:
Hi,
I own a "Retroflag Classic USB Gamepad" with a Genesis layout (6 buttons and 2 shoulder buttons) and found this issue. It shows exactly the same behaviour as the SNES variant. With the attached firmware, it is recognised by MiST after a reset. It then works fine.

Thank you for solving this problem and I hope the change enters the official firmware soon.

gazsp, 2021-09-27:
@gyurco The pad now works fine with the latest firmware. Thank you!

gyurco, 2021-10-17:
I'm closing this as it works - with the small inconvenience with the extra reset required.

closed 2021-10-17 by gyurco

gyurco, 2021-11-21:
The new firmware supports this gamepad in XBOX mode (called Direct Input in some manuals). It works without an initial reset for me.