mist-devel/mist-binaries #110: Amstrad_alt - Led Storm graphic bug

AxelStone, 2021-06-15:
Good day, I've seen an error with game LED Storm that can affect to other games. When you go slow with the car, the scroll does strange jumps up - bottom instead of a normal scroll.

Thanks so much for your work!

gyurco, 2021-12-10:
I've re-checked how the scrolling is done in LED Storm. It simply uses R5 to add extra scanlines to the frame, and variable frame lengths are not always tolerated by displays. My VGA display just goes blank, but my Samsung LCD TV via SCART displays the game properly. I don't think anything can be done in the core.
Is the game on the same display on a real CPC works correctly?

AxelStone, 2021-12-23:
Hello @gyurco sorry for delay, you are right, after some checks and ask CPC comunity it seems that the trick depends on the original CPC screen, so it's something related to the screen itself not the machine.

Thanks for your support.

closed 2022-01-17 by gyurco