mist-devel/mist-binaries #112: time for usbrtc is not set.

xdemax, 2021-07-18:
hello! can't set clock time in emu-tos and amiga os. latest firmware, latest kernels.

gyurco, 2021-07-18:
Only the original ST core can set the clock, all others are read-only currently.

xdemax, 2021-07-18:
thanks, got it, installed the old version of the ST kernel, the time is saved, but the date is not.

gyurco, 2021-07-18:
It depends on the XControl module you're using, I don't remember which one I've used, but definitely worked. Much easier to connect the USB-RTC to a Linux, and set it up via hwclock.

xdemax, 2021-07-19:
i came to roll back to the firmware from 03/03/19, set the clock and date in the 'atari tos', pull out the attiny, flash the latest firmware and return the attiny to their place. now both the clock and date are working.

gyurco, 2021-11-27:
In recent firmwares, it's also possible to set it in the OSD.

closed 2021-11-27 by gyurco