mist-devel/mist-binaries #113: Firmware/core hangs when copying large files to Minimig-AGA HDF file

boomlinde, 2021-07-24:
I moved archives from the SD card (mounted via the SD0 mount from MinimigUtils) to RAM: where I extracted them. This worked with no problems. However, when I copy the files from RAM to a partition in the HDF file, the MiST hangs entirely and gets stuck with the disk light consistently on. The menu can't be activated with the keyboard, nor the button on the front panel. It is still possible to reset the MiST from button on the left side of the front panel.

There is some initial disk activity indicated by the flashing red light, but within about a second the hanging seems to occur at an inopportune moment, so it seems to always result in some corruption of the file system.

I can consistently reproduce by e.g. copying doom.wad from RAM: to my work partition in the HDF. Smaller files work well. For example, I managed to copy Xvi from Aminet (which only contains a few, small files) without issue in the manner described above.

The Minimig core version is 21.6.18. MiST ARM/FILE are ATH210618. My SD card is FAT32 formatted. No turbo options are active. Everything works well with the other cores (though I rarely write data with the other cores) and the Minimig core works well until I encounter this problem.

I recently updated the firmware and core for the first time in a few months. Could this have something to do with the new ExFAT driver?

boomlinde, 2021-07-25:
Here is an interesting development. When copying to disk, I used the GUI drag/drop method. This still consistently hangs (now trying with the latest firmware and minimig core). However, when I use the CLI to copy the file (`COPY ram:doom.wad work:games/Doom`), it does not hang.

Perhaps there is a difference between how COPY in AmigaDOS and GUI copy works. Regardless of what might be wrong on the Amiga software side (it did work previously though) I think it shouldn't result in the MiST completely hanging with no OSD available. The core is still generating video, but does not react to input, and there is no way to get to the OSD.

Now, I almost always do disk -> RAM -> disk copies out of habit (to extract lha archives in the RAM disk). Perhaps direct disk -> disk copies provide a natural bottleneck that prevents this issue from occurring in that case, and the default number of buffers for the COPY command might be different from GUI copy. I will try disk to disk copies tomorrow. I also want to try downgrading the FW to the last non-exFAT version to see if that helps.

On the Amiga side I am using KS 40.68 and WB 40.42. I am willing to upload and share the HDF and Kickstart I use over email (for testing purposes only) if someone would like to troubleshoot this.

gyurco, 2021-07-25:
ARM should not freeze from the core. Maybe there's a filesystem corruption which confuses FatFS. Can you try chkdsk/fsck/scandisk/whatever on your SD Card?
Also trying an older firmware could help to discover where is the issue.

robinsonb5, 2021-07-25:
The symptom (works with small files, doesn't work with large files) sounds a lot like the old MaxTransfer problem.  Can you check, using HDToolBox, that the MaxTransfer parameter for all partitions is set to no more than 0x1fe00.

I'm not sure why it's causing the firmware to hang - unless the MaxTransfer bug's causing the firmware to request a non-existent block from the HDF, and the firmware's not gracefully handling the error condition?

boomlinde, 2021-07-25:
I just ran fsck on the SD card. Some dirty bits set, wrong cluster summary, so I'll give it a try with a different, freshly formatted SD card. Changing the MaxTransfer size seems like an effective workaround, but was not necessary before. While copying the core and menu are still unresponsive for pretty long stretches of time (seconds at a time), but that might be normal.

Hopefully I have time tomorrow night to walk back through the firmware releases as well. Thanks for the help so far and for suggesting possible causes!

gyurco, 2021-07-25:
Stalling for several seconds means a very big (or slow) transfer. What's the cluster size on the SD card?
The MaxTransfer bug might be related to LBA mode - what CHS values should be stored after a transfer in this case?

boomlinde, 2021-07-25:
The cluster size is 8192.

robinsonb5, 2021-07-25:
> Stalling for several seconds means a very big (or slow) transfer.

Or a distant seek with a failed index? (Assuming we're talking about FastFileSystem here, the directory blocks are midway through the partition, so creating files will involve writes to the beginning for the data, then seeking to the midpoint to update the directory contents.)

> The MaxTransfer bug might be related to LBA mode - what CHS values should be stored after a transfer in this case?

I don't know, but would a Kickstart 3.1 installation with scsi.device even be using LBA mode?

gyurco, 2021-07-26:
> Or a distant seek with a failed index? (Assuming we're talking about FastFileSystem here, the directory blocks are midway through the partition, so creating files will involve writes to the [...]
That's plausible, with a 8k cluster size.

> > The MaxTransfer bug might be related to LBA mode - what CHS values should be stored after a transfer in this case?
> 
> I don't know, but would a Kickstart 3.1 installation with scsi.device even be using LBA mode?
True, probably not.

gyurco, 2021-07-26:
> The cluster size is 8192.

That's too small, 32K or 64K are the recommended values, especially with big HDF files.

boomlinde, 2021-07-26:
I reformatted the card with 32k clusters. This seems to have had a drastic positive effect on overall disk performance, but now the problem occurs while copying large files again, whether I've limited the MaxTransfer size to 0x1fe00 or not.

I will now try moving back in the firmware history to see if the problem can be tied to a specific release.

boomlinde, 2021-07-26:
Another interesting finding:

I downgraded the firmware to 210330 and ran the last released minimig core released at that time (minimig_mist_rtg_201122.rbf). The problem did not occur in this configuration. Then, I tried that older firmware with the newest minimig-aga release. The problem did occur in this combination. Finally, I tried minimig-aga 201122 with the latest firmware. The problem does not occur here.

I'm not setting MaxTransfer to 0x1fe00 in any of these test cases, but leave it with all bits set.

This leads me to believe that the issue pertains somehow to changes in the current release of the Minimig core rather than changes in just the firmware. I can't find a change log or the source code for the Minimig core, so I'm not sure what I'm missing out on by using an earlier version of the core.

gyurco, 2021-07-26:
Minimig-AGA sources are here:
https://github.com/minimig-dev
If you're familiar with git bisect, you can pinpoint the commit which introduced the issue. Or you can try to revert the  IDE changes around last November-December, and see if it helps (there were 3 IDE-related commits).

gyurco, 2021-07-26:
Looks like I could reproduce the issue: it hangs when the count is 256:
HDD: IDE0: write CHS, 16580.0.4:530563, 256

gyurco, 2021-07-26:
@robinsonb5  Maybe this one?
```
assign drq = (fifo_full & pio_in) | (~fifo_full & pio_out & sector_count != 0); // HDD data request status bit
```
If the sector count is 0, DRQ never raised. If it's the problem, I wonder how to differentiate 256 with 0 then. Maybe at write, if 0 is written, then it should be converted to 256?

boomlinde, 2021-07-26:
Here is some relevant information I think: http://www.lysator.liu.se/~bonte/text/ata-r32.txt (7.2.11)

gyurco, 2021-07-26:
Can you try this build?
[minimig_mist.zip](https://github.com/mist-devel/mist-binaries/files/6877760/minimig_mist.zip)

boomlinde, 2021-07-26:
Sure!

robinsonb5, 2021-07-26:
>  Maybe at write, if 0 is written, then it should be converted to 256?

Probably, yes - sector count of 0 represents 256 sectors both for reads and writes.  I'm not sure how sector count 256 (0) can happen if MaxTransfer's set to 0x1fe00 though, since that's 255 sectors.

boomlinde, 2021-07-26:
That build doesn't seem to want to boot from my hdf at all unfortunately. I can see some disk activity, but I never end up at the workbench so I guess there is some corruption.

@robinsonb5 Yes, I noted later above that I probably screwed up my 32k sector + 0x1fe00 MaxTransfer test. I tried it again and it works as long as I limit the MaxTransfer.

gyurco, 2021-07-26:
> That build doesn't seem to want to boot from my hdf at all unfortunately. I can see some disk activity, but I never end up at the workbench so I guess there is some corruption. The MiST firmware functions still seem to work after this happens, though.

Even if you give it some more time?

boomlinde, 2021-07-26:
I probably gave it a minute or two, long after disk activity ceased anyway. I'm letting it run a while longer now though while I grab something to eat.

gyurco, 2021-07-26:
Sorry, I've uploaded wrong version. Replaced the zip, try that please.

boomlinde, 2021-07-26:
I've tried the new RBF now. This seems to work well with both MaxTransfer 0xffffff and 0x1fe00, and I have verified the integrity of the file after copying using md5sum, so no silent corruption going on either. So, so far it's very promising. If there are other tests you want me to do, let me know. The file I copied this time was again doom.wad at ~12MB. I'll put it through some more normal use tonight as well.

I will comment again if I encounter any issues with this version, but go ahead and close this when you feel it's appropriate.

Cool to see this addressed so quickly. :+1:

boomlinde, 2021-07-29:
Closing this. I have used the fixed version above for quite a bit of copying in the past few days, and can confirm that minimig_mist_rtg_210727.rbf also works.

closed 2021-07-29 by boomlinde