mist-devel/mist-binaries #115: Don't work USB Keybord.

xolod79, 2021-08-23:
**Between releases 20210530 (last work) and 20210618 my keyboard stopped working.
If I switch to another USB socket, it starts working until it is powered off by MiST**
Console log on 20210618:

Minimig by Dennis van Weeren                                                    
ARM Controller by Jakub Bednarski                                               
                                                                                
Version ATH210618                                                               
                                                                                
SDHC card detected                                                              
spiclk: 24 MHz                                                                  
usb_init                                                                        
max3421e_init()                                                                 
Chip revision: 13                                                               
busprobe()                                                                      
usb_reset_state                                                                 
Partition Count: 1                                                              
Partition: 0  Start: 2048  Size: 61829120                                       
Partition type: EXFAT                                                           
fat_size: 7552                                                                  
fat_number: 1                                                                   
fat_start: 4096                                                                 
root_directory_start: 7                                                         
dir_entries: 0                                                                  
data_start: 12288                                                               
cluster_size: 64                                                                
free_clusters: 4294967295                                                       
ChangeDirectoryName: / -> / = /                                                 
debug_mode = 0                                                                  
Error opening file "/MIST.CFG" (4)                                              
INI_PARSER : Start INI parser for core "(null)".                                
INI_PARSER : Can't open file CORE    ARC !                                      
ARC CONF STR:                                                                   
loaded_from_usb = 0                                                             
FPGA bitstream file opened, file size = 217507                                  
[**************]                                                                
FPGA bitstream loaded                                                           
FPGA configured in 1304 ms                                                      
ident = a4                                                                      
Identified 8BIT core                                                            
Sending core mod = 0                                                            
Core name from FPGA is "MENU"                                                   
Loading config /MENU.CFG                                                        
Looking for MENU.ROM                                                            
Looking for /MENU.ROM                                                           
Looking for /MENU.RAM                                                           
Looking for /MENU.VHD                                                           
error mounting /MENU.VHD (4)                                                    
Looking for /MENU.HD0                                                           
error mounting /MENU.HD0 (4)                                                    
Looking for /MENU.HD1                                                           
error mounting /MENU.HD1 (4)                                                    
Looking for /MENU.HD2                                                           
error mounting /MENU.HD2 (4)                                                    
Looking for /MENU.HD3                                                           
error mounting /MENU.HD3 (4)                                                    
INI_PARSER : Start INI parser for core "MENU".                                  
INI_PARSER : Opened file /MIST.INI with size 1330 bytes.                        
INI_PARSER : line(0): "[mist]".                                                 
INI_PARSER : Got SECTION 'MIST' with ID 1                                       
INI_PARSER : line(0): "scandoubler_disable=0".                                  
INI_PARSER : Got VAR 'SCANDOUBLER_DISABLE' with VALUE 0                         
INI_PARSER : line(0): "csync_disable=1".                                        
INI_PARSER : Got VAR 'CSYNC_DISABLE' with VALUE 1                               
INI_PARSER : line(0): "mouse_boot_mode=0".                                      
INI_PARSER : Got VAR 'MOUSE_BOOT_MODE' with VALUE 0                             
INI_PARSER : line(0): "key_menu_as_rgui=0".                                     
INI_PARSER : Got VAR 'KEY_MENU_AS_RGUI' with VALUE 0                            
INI_PARSER : line(0): "joystick_emu_fixed_index=1".                             
INI_PARSER : Got VAR 'JOYSTICK_EMU_FIXED_INDEX' with VALUE 1                    
[... 111 lines cut ...]
0010: 75 01 95 03 81 02 95 05 81 01 c0 05 09 09 01 a1   u.•..•..À....¡        
0020: 01 85 03 0a 8a 01 0a 23 02 0a 2a 02 0a 21 02 0a   .….....#..*..!..        
0030: 27 02 0a 26 02 0a 25 02 0a 24 02 0a 83 01 0a cd   '..&..%..$..ƒ..Í        
0040: 00 0a b7 00 0a b6 00 0a b5 00 0a e9 00 0a ea 00   ..·..¶..µ..é..ê.        
0050: 0a e2 00 0a 94 01 0a 92 01 0a 34 00 0a b2 00 0a   .â..”..’..4..²..        
0060: b4 00 0a 06 02 0a b8 00 0a 3d 02 15 00 25 01 95   ´.....¸..=...%.•        
0070: 18 75 01 81 02 c0 05 09 09 01 a1 01 85 04 0a 56   .u..À....¡.…..V        
0080: 01 0a 57 01 0a 58 01 0a 59 01 0a 5a 01 0a 5b 01   ..W..X..Y..Z..[.        
0090: 15 00 25 01 95 06 75 01 81 02 95 01 75 02 81 01   ..%.•.u..•.u..        
00a0: c0                                                À                       
HID NONE: report type = 0, size = 1                                             
HID configured                                                                  
 -> accepted :-)                                                                
status change on port 1, 0x110100                                               
Status of port 1:                                                               
 powered                                                                        
Changes on port 1:                                                              
 connected                                                                      
 reset                                                                          
status change on port 1, 0x110100                                               
Status of port 1:                                                               
 powered                                                                        
Changes on port 1:                                                              
 connected                                                                      
 reset

gyurco, 2021-08-23:
It doesn't seem to be a firmware issue:
```
port 3 reset complete!
usb_configure(parent=1 port=3 lowspeed=1)
using free entry at 1
Setting addr 2
usb_set_addr(new=2)
failed to assign address
```
USB protocol error somewhere. If you can build a new firmware, you can investigate the error code, just change
```
puts("failed to assign address");
```
to
```
iprintf("failed to assign address (rcode=%d)", rcode);
```
in usb.c

Then the rcode can tell something about the problem.

BTW, other devices are working in port3?
And after you plug your keyboard to port2, what are those events on port1?
Can be a powering issue.

xolod79, 2022-02-17:
I have a USB-RTC connected to port 3 of the USB.
Power problems cannot be ruled out. But on firmware before 30.05.2021(including it) everything works fine.

retrofun, 2022-06-09:
@xolod79: is this still an issue with the current firmware?

USB enumeration was tweaked in the meantime. Does it work with firmware 220220+?

xolod79, 2022-06-09:
I confirm the latest firmware working normally with the keyboard

closed 2022-06-09 by xolod79