mist-devel/mist-binaries #116: C64 Tape bug??

Julitium, 2021-09-07:
Hi. 

I have detected a bug (I Think) in certain games related to the operation of the tape on C64 Core.
When it finishes loading the main program and asks to change to side B and press play. This is not the case, when side B is loaded from the OSD it does so with play activated (that is indicated by the middle LED) but the software does not detect that "play" and you have to press play / stop several times.
Detected in:
Flimbos Quest.
Hawkeye.
I hope it is an easy thing to correct (Maybe Motor On and Play signals are mixed ?? .... No idea).

Thank you so much for your attention.

All the best.

gyurco, 2021-09-08:
Flimbos Quest was working before. I think it expects two change, one rewind button and one play. So if you press play/stop twice, it'll work. That's normal.

Julitium, 2021-09-09:
Hi.

Ok. Maybe that's it.

Thanks.

closed 2021-09-09 by Julitium