mist-devel/mist-binaries #117: PC-Engine: YPbPr Output broken since release 210705

uchristo, 2021-09-08:
Purple output since Release 210705. OSD is not/only partially affected. See Pictures in  the following post

https://www.atari-forum.com/viewtopic.php?p=421857#p421857

uchristo, 2021-09-08:
This one seems to be the relevant commit:
https://github.com/mist-devel/TurboGrafx16_FPGA/commit/d3791e5c5ed23321bf1b30748cbda82d24c9797f

"added verilog rgbtoyprpb"

robinsonb5, 2021-09-08:
I think this is just a missing #(WIDTH(6)) in the instantiation - could you test this version, please?  (I don't have a Component-capable monitor for testing.)
[tgfx16_mist.zip](https://github.com/mist-devel/mist-binaries/files/7130758/tgfx16_mist.zip)

uchristo, 2021-09-09:
I can confirm this fixes the issue! Thank you very much for this very quick response!

closed 2021-09-09 by uchristo