mist-devel/mist-binaries #123: Control issues with Cadaver (minimig-aga)

sprudel79, 2021-11-29:
Hi everyone,
I am using the MIST with latest firmware (firmware_211121.bin) and the core minimig-aga (minimig_mist_rtg_210727.rbf) and have a problem with one of my favorite games which is Cadaver.
I have tested two configurations, I got exactly the same issues:
1. My preferred way via HDF and WHDLoad (Amiga 1200 based config with rom 3.1)
2. Loading the game directly via ADF with a Amiga 500 based config and rom 1.3

I should say I am using a DB9 connected Joystick called "ArcadeR" (https://retroradionics.co.uk/#!/ArcadeR-9-pin-ATARI-standard-Joystick/p/168982750/category=0)
The joystick itself is correctly detected by the MIST and I can check it successfully to see that all directions incl. the two fire buttons are working.

So what's the issue now?
When the game starts I can use the Joystick only to the left and right but the character doesn't move for top or down movements with the Joystick.
Moving the Joystick downwards will even make the character jump! So super strange...
I should say that some other games are working so it might be something special that the core doesn't yet support?

I have btw. checked my setup with HDF and the same joystick in an emulator (shame on me) where it worked properly.

Any ideas from anyone?
Thanks in advance.

harbaum, 2021-11-29:
Chances are the game does something unusual like reading an undocumented register mirror or the like.

It might be worth searching for "cadaver" in emulator source code. Often when emulators contain code for such edge cases the author's add a comment that indicates what this code is for.

sprudel79, 2021-11-29:
Thanks @harbaum for your reply.
Good idea to check the source code. The emulator I was using is the PUAE Libreto one which is part of Retroarch. The source code is available here: https://github.com/libretro/libretro-uae
I have quickly checked out the entire source and actually found some custom code for certain games but unfortunately nothing regarding Cadaver.

gyurco, 2021-11-29:
It works for me. Maybe you can try to switch CD32PAD on/off.

sprudel79, 2021-11-30:
I have to apologize, it's working now also for my side.
I had some fancy Joystick mapping options in my mist.ini, once I had removed them it worked like a charm.
I had relied on the following mist.ini:
https://github.com/mist-devel/mist-binaries/blob/master/cores/mist.ini
which I had used as a basis.

Here's my original mist.ini:
```
[mist]
ypbpr=0
scandoubler_disable=1          ; set to 1 to run supported cores in 15khz
forced_scandoubler=0
composite_sync=1
mouse_boot_mode=0              ; set to 1 if a mouse does not work well
joystick_disable_shortcuts=0   ; set to 1 to remove joystick -> keyboard commands 
joystick_ignore_hat=0          ; set to 1 if having issues on gamepads with 'POV hat'
joystick_ignore_osd=0          ; set to 1 to prevent gamepad from controlling OSD
key_menu_as_rgui=0             ; set to 1 to bind Menu USB key to RGUI (=Right Amiga in Minimig)
joystick_remap=0583,2060,1,2,4,8,10,20,100,200,400,800,40,80
joystick_remap=0F0D,00C1,1,2,4,8,200,20,10,100,400,800,400,40,800,80
joystick_remap=0079,18D2,1,2,4,8,100,10,20,200,400,800,1000,2000,40,80
joy_key_map=1000,E2
joy_key_map=2000,E6
joy_key_map=4,51
joy_key_map=8,52
key_remap=53,2F
key_remap=47,30
[snes]
joy_key_map=1000,45
```

Now I just have:
```
[mist]
ypbpr=0
scandoubler_disable=1          ; set to 1 to run supported cores in 15khz
forced_scandoubler=0
composite_sync=1
mouse_boot_mode=0              ; set to 1 if a mouse does not work well
joystick_disable_shortcuts=0   ; set to 1 to remove joystick -> keyboard commands 
joystick_ignore_hat=0          ; set to 1 if having issues on gamepads with 'POV hat'
joystick_ignore_osd=0          ; set to 1 to prevent gamepad from controlling OSD
key_menu_as_rgui=0             ; set to 1 to bind Menu USB key to RGUI (=Right Amiga in Minimig)
```
and Cadaver works properly.
Thanks again everyone for your support!

closed 2021-11-30 by sprudel79