mist-devel/mist-binaries #128: I found an strange behaviour in FPGAGEN

PepAlacant, 2022-05-05:
I noticed 2 things. There are times that the rom is not loaded, then... if reset the MiST and then start FPGAGEN, the game that I previously tried to load appears automatically.
Another thing I found is a very strange behaviour in the game Mamono Hunter Yohko. The game has a lot of collision issules and invisible obstacles. All can become very buggy. When the game runs in "CPU Overclock" the things seems to work well.

I'm using a Manuferhi's MiSTICA FPGA.

Grettings !

gyurco, 2022-05-06:
This game was mentioned in the thread:
http://gendev.spritesmind.net/forum/viewtopic.php?f=24&t=347
Generally it writes to the main RAM with the Z80, which is not supposed to be work on original HW. But in some circumstances, it does.

PepAlacant, 2022-06-13:
This core has strange behaviour. Some strange things, like screen desync and other problems if the CPU is not in Turbo mode.

gyurco, 2022-07-06:
Try the core posted here:
https://github.com/mist-devel/fpgagen/issues/71

closed 2023-04-26 by gyurco