mist-devel/mist-binaries #130: c16 - core and new game Lykia

micv2, 2022-08-07:
Hello, 

the new game Lykia for the Plus/4 causes the following problems on the MiST FPGA (C16 core). 

1) The SID sound is awful, it cuts out and distorts. 
2) Beginning of the game is ok, title picture with menu is also ok. But then in the game my Eizo (50Hz) monitor shows an error 49Hz, why? See image. 
3) It's an awesome game with 121 colors and SID sound support, too bad georam isn't supported by C16-core. 

Best regards, 
MiC

https://puls4r.de/portfolio/lykia/
http://plus4world.powweb.com/software/Lykia

[image]

[image]

gyurco, 2022-08-08:
Hi,

1. Both SIDs?
2. If you load a PAL kernal, can your display handle it?
3. I thought georam is C64 specific...ahh, it's neoram. Theoretically it's not hard to support it. But in this game, it's only replaces the disk loading, or does it add extra features?

micv2, 2022-08-08:
Here is the SID sound via sidekick264, sounds clean..

https://youtu.be/wRnbbfyBtpA

robinsonb5, 2022-08-08:
I just tried it on my port to the TC64 (which has the MiSTer SID) and the result is the same.  It sounds almost like audio buffer underruns, which is ridiculous because there are no audio buffers!

gyurco, 2022-08-08:
The C64 version was good when I tried, and it has the same SID modules.

Julitium, 2022-08-08:
Hi.
I have also tried the game.
It seems that it is associated with sustain or (sure) a possible filter effect.
It should be noted that the implementation of the SID filter is based on tables (not real time), so it is acceptable considering these details.
Apparently the bug is less evident in 8580 mode (this is normal, AMP is less saturated).
All the best

micv2, 2022-08-08:
Hi,
found out something, i choose the disk version without SID so i have no picture problem. If I select the disk SID version, the monitor shows 49Hz. The SID support does something with the image timing?

Best regards,
MiC

[image]

[image]