mist-devel/mist-binaries #132: Atari 800 Core and new "Rewind" Demo

micv2, 2022-11-24:
Hello, 

I can't get this new Rewind (1MB) demo to start, what am I doing wrong?
Best regards,
MiC

https://demozoo.org/productions/310140/
https://www.pouet.net/prod.php?which=91795
https://youtu.be/3s2vqN1jWgU

gyurco, 2022-11-24:
As I see, only a big XEX file is available for this demo. Try to get a real disk image (like ATR) and see if it works.

micv2, 2022-11-24:
Hi, 
thanks for the answer. Infuriating I have two Atari XL/XE, both only have 512K. MiST was my hope to start the demo. There is probably no ATR. Found a thread where another XEX was attached. Best regards, MiC

https://forums.atariage.com/topic/337943-rewind-new-demo-by-ng-and-zelax/

[rew_pak.zip](https://github.com/mist-devel/mist-binaries/files/10086982/rew_pak.zip)

gyurco, 2022-11-25:
There's an epilogue posted on the same thread in Atari Age, which works if you select 1088k, but crashes similarly as the main demo in any 576k mode. So I suspect the bank switching implementation has some incompatibility with the demo. Maybe you can ask Foft on Atari Age (original author of the core) if he has an idea?

micv2, 2022-11-25:
This epilogue works on my real Ataris with 576K. I'm already whining in the forum because of the 1MB demo. Perhaps someone can split it in two parts. Thank you.

https://forums.atariage.com/topic/337943-rewind-new-demo-by-ng-and-zelax/page/2/#comment-5160817

gyurco, 2022-11-25:
What I try to explain is the epilogue doesn't work in the core with 576K setting. So maybe it's some general incompatibility with the bank switching method.

micv2, 2022-11-25:
I now understand what you mean. Thanks gyurco for taking care of it 👍