mist-devel/mist-binaries #134: Last Mist firmware bug ?

tricipi, 2022-12-07:
Hello,

Once updated my Mist (1.4) with new firmware, I found the following behaviour with Minimig core :  every time I pull the DB9 joystick connected to port #1 to right position, the video signal is forced a "reset" (NO INPUT SIGNAL ). I have not yet tried with previous firmware, but restoring the old "factory" firmware present (161229) this problem is not present.

By
tricipi

gyurco, 2022-12-07:
I cannot imagine how the DB9 ports can affect the video output. Probably only if your joystick short-circuits the +5V line if you move it to the right.

tricipi, 2022-12-07:
Ok, but it happens only with lastest firmware update. The 200622 version is not affected. Meanwhile I try to find the "last" firmware version without this problem

tricipi, 2022-12-07:
Furthermore, It happens with various DB9 joystick, not only one, and this does not happens with other cores (i.e. th C64 one)

gyurco, 2022-12-07:
Is this happens with the port used normally for the mouse on Amiga? That might confuse the Amiga HW/Amiga OS, but I still don't see how the firmware would be connected to this. The video output is totally independent of the ARM CPU (it's on the FPGA).

gyurco, 2022-12-07:
Actually I'm not sure if the latest Minimig-AGA core fully works with such an old firmware.

harbaum, 2022-12-07:
Does this happen with a simple Atari style joystick? If not being touched these don't create any connection at the port at all. There is no electrical difference between no joystick connected and a simple classic joystick connected in center position and fire button unpressed.

This sounds to me like you have some bad solder joint in your MIST or the like. I cannot imagine how this should relate to the firmware.