mist-devel/mist-binaries #140: Video mode hopping issue with MENU core

egrath, 2023-04-22:
**Issue:**
When the scandoubler is disabled (scandoubler_disable=1) the MENU core hops between different video output modes continiously. It seems that some (or even more, most) TVs have no problem with this [1], the Retrotink 5X upscaler complains about this and has issues displaying the video output of the core correctly - the entire image is wobbling around like its completely sloshed.

The two modes it oscillates between are 312p and 625i - at least the Retrotink shows this as the modes. The frequency of the oscillation is around 1/5 second.

Resolutions while in the MENU core:
[image]
[image]

Other cores don't share the same behavior, they output a nice and stable video signal:
[image]
[image]

**How to reproduce:**
Don't know for sure, but probably with some kind of device which shows the mode of the current input video signal...

**Expected behavior:**
Same stable video signal as other cores

**Additional notes:**
Can't tell if this behavior on the MENU core only occurs when the scandoubler is disabled as the Retrotink has no VGA input.

gyurco, 2023-04-25:
> 
> **Expected behavior:** Same stable video signal as other cores
> 
As there are no two cores with identical sync timings, it's hard to tell which one should be the base.
Also currently I don't have any 15kHz displays to test. But it's easy to do some changes here:
https://github.com/mist-devel/Menu_MIST/blob/master/menu.sv#L208

Maybe vsync should be aligned to hsync, e.g. change vsync generation to:

```
	if(vc == line_max-3 && hc == 535) VSync <= 1;
		else if (vc == 0 && hc == 567) VSync <= 0;
```

And see what happens.

egrath, 2023-04-26:
Success!

I've managed to build a new MENU.RBF from source with the modifications you told in the last post and the 15 khz signal is picked up correctly by the RetroTink and also by my two TVs i've also used to test the modification.

Attached the MENU.RBF (renamed as CORE.RBF) in case anyone else is willing to test if it has any side effects i did not noticed.

[image]
[CORE.zip](https://github.com/mist-devel/mist-binaries/files/11335693/CORE.zip)

gyurco, 2023-04-26:
Good! I added the change to the git repo.

egrath, 2023-04-26:
Perfect, fixed with da7e4fc5fa9468a34b1c37d4e213a8d220a67fbf

egrath, 2023-04-26:
For reference: https://github.com/mist-devel/Menu_MIST/commit/da7e4fc5fa9468a34b1c37d4e213a8d220a67fbf