mist-devel/mist-binaries #141: Mistery core and ACSI hard disc support

micv2, 2023-05-18:
Hallo,
I created a disk image for TC64 (Mistery core). There are two partitions witch 2x512MB, and the extension of the image is VHD. I can select and boot it with Mistery core of TC64, it works very well. Now I wanted to use the image on the MiST Mistery core, but the VHD image cannot be selected. It says under harddrives, disk not found. Which extension must my image have, IMG? HD? Please help. 
Best regards, MiC 👍

harbaum, 2023-05-18:
I've just heard that Lotharek gives full support for the hardware and software.  Please contact him via his shop. He'll hopefully be giving you all the support you need.

You should in general always contact the hardware vendors first if you have questions or need support.

micv2, 2023-05-18:
Thanks for the answer. But what does the Lotharek have to do with the Mistery core? Gyurco was always helpful 👍😎

gyurco, 2023-05-18:
BTW, answer to the question is in the Wiki (anyone reads that?)
https://github.com/mist-devel/mist-board/wiki/AtariMassStorage

micv2, 2023-05-18:
I got it, the extension must be .hd. So example "image.hd" . So it's different than on the TC64. Mistery core, but everyone seasons the soup differently. Too bad it isn't in the description. Greetings and thanks to all 👍😘

micv2, 2023-05-18:
> Once I read from Lotharek that he sucks in software :)

He likes beer 😎

harbaum, 2023-05-18:
Those making money should imho also give some support. He wouldn't be selling anything without many other people doing the hard work. 

Maybe he'll help. Who knows ...

gyurco, 2023-05-18:
I'm sure developing cores is not amongst his abilities.

I wonder how many new users will choose the new MiST 1.5, I only heard about people who bought it for a second device. But if only one guy starts to do some development, then it's a win.

closed 2023-06-10 by gyurco