mist-devel/mist-binaries #144: [PC XT] Writing to disk destroys filesystem on SD card

famaniel, 2023-07-27:
I've been trying the MIST port for [PC XT](https://github.com/mist-devel/mist-binaries/tree/master/cores/pcxt). following the guidelines in the [PC XT Demistify Wiki](https://github.com/somhi/PCXT_DeMiSTifyl). Setting up the core and the files works perfectly, and I could boot into DOS after a few attempts. 

However, whenever I write a file into the HD image, or just save the settings in the OSD, the filesystem on the SD Card (16MiB SanDisk) is destroyed, and the system crashes. I tried a differend SD card with the same result. 

I'm using the core on MIST 1.3 with  firmware_230705. The other cores I'm using work fine. 
Anyone else experienced this?

Thanks

gyurco, 2023-07-28:
It's rather strange. I never encountered such issue (MiST destroys the filesystem on the SD card). Are you using exFAT?

famaniel, 2023-07-28:
No, I formatted the SD card with mkfs.vfat /dev/sdx1 (where sdx ist the normal device Id). I tried one "Platium" card, and one from SanDisk. Both 16GiB SDHC. Maybe it has something to do with that.

gyurco, 2023-07-28:
It's hard to tell. Minimig/Archie/CPC (with SymbOS) use the same IDE code, are they working properly?

famaniel, 2023-07-29:
I ran a series of experiments: I have an SD card I've been using for years for the minimig. It always worked fine, and still does. I reformatted both SD-Cards, and copied the minimig installation to both of them. They did boot up, but at the first attempt of writing to the hard drive image or a virtual diskette, AmigaDOS threw an I/O error, and crashed. The problem was the same as before. This happend with both cards. So today I hit the shops, and bought a bunch of SanDisk branded 32 GiB SDHC cards. I tried them out, including all sorts of I/O operations in DOS, and they work fine. Also tried the minimig with no problems. So it appears, that the problem is in my SD-cards, and not in the code. I'm beginning to suspect the previous SanDisk-branded card is actually counterfeit. But I cannot check that, since my SD card reader doesn't allow getting the CID. 

Thanks so much for your help! I'd say my problem is solved.

closed 2023-07-29 by famaniel