mist-devel/mist-binaries #145: spectrum.rom

thecaver99, 2023-08-13:
very difficult to create new spectum rom.
can we just not load the roms from with in the core.

gyurco, 2023-08-14:
You mean load ROM from the OSD? How will it help to create a new ROM?

thecaver99, 2023-09-29:
Sorry for late replay, if was able to load from osd then wouldn't need spectrum.rom,
I've tried to make a new rom file and got most things to work except the multiface part.
If things could be put in the osd like which rom to load and add ons like floppy drive,, multiface very similar to like amiga, atari st, c65 etc it would make the core alot more customizable for end user

gyurco, 2023-09-30:
But for adding a load option for each component, 10+ options would be needed in the OSD. Or did you think about something else?

thecaver99, 2023-09-30:
Ah right ok I didn't realise it would be so many different things there, 

Would it not be easier just have a roms folder and have the revelven roms with correct names in there for it to read, so if you say wanted the gw.rom in place of the 48k rom we could just rename it, or having the plus 3e in place of the normal plus 3 we just rename the roms to match to save making the spectrum.rom or even a tool to put the roms in the correct place in the spectrum rom file, so it would be easier to change as needed?

gyurco, 2023-09-30:
That would be a firmware feature, the core just need to receive the whole ROM together.
However there's already the rom= option in mist.ini, which is similar what you need:
https://github.com/mist-devel/mist-board/wiki/DocIni#rom

You can add this option like this:
```
[spectrum]
rom=01GLUK
rom=02TRDOS
rom=03SEBAS4
rom=05P3EMMC
rom=09PLUSD
rom=10MF128
rom=11MF3
rom=12PRETTY
```

thecaver99, 2024-06-14:
thank you, i will give this a try.