mist-devel/mist-binaries #146: firmware_240105.upg bricked my MIST

mincebert, 2024-01-08:
This might be just a one-off or I might have done something wrong, but I copied the [firmware_240105.upg](https://github.com/mist-devel/mist-binaries/blob/master/firmware/firmware_240105.upg) file as `firmware.upd` onto an SD card, picked 'Upgrade firmware' but it never completed after 10-20 minutes and had to powercycle the MIST.  Now it doesn't boot.

I'm sure I'll be able to recover it through the ARM JTAG port (I haven't had time yet), but I thought I'd just report this in case it's a more general problem!  When I do try that, I'll try putting the new firmware on directly and downgrade again, if that doesn't work.

gyurco, 2024-01-08:
https://github.com/mist-devel/mist-binaries/tree/master/firmware#important-note

closed 2024-01-08 by gyurco

mincebert, 2024-01-08:
Thanks for pointing me at that — I tend to get the updates by doing a `git pull` and so missed the update in the note.  I also didn't check what the meaning of the intermediate file was!

I'll be more careful next time!