mist-devel/mist-binaries #147: Firmware 240105 MiniMig Issue - Possible ?

Torti-the-Smurf, 2024-01-11:
Hi, my name is Torti and i am new here on GitHub.

Anyway; i do LOTS of testing for the Minimig Core , especially for Robinson, 
but i have a question about the latest Firmware if you dont mind.

If its possible that it broke some stuff in the MiniMig Core ?
The game Clown-O-Mania hangs now.    (T_T)

Right at the start in the STARBYTE LOGO... it hangs on a very high pitched note and freezes...
Never happend before.

I did not changed anything... thats so weird. Is that some "bad influence" from the new FW ?

Oh and one more thing  ... if you dont mind.
It would be really neat if that Joystick OSD "stuck Bug would be fixed".....

You know the one, where you control the game and keep holding a direction and then hit F12 for the OSD.
( i am using a 3 Button Mega Drive joypad .. .. it has such a great D-PAD - my Amiga Pad of choice)

When this happends, the OSD gets stuck and nothing besides a power off can be done. 
Not that scenario happends often, but when it does for some reason, (mostly by exident)  it pretty annoying ;)

In any case, HUGE thanks and RESPECT  for supporting the MIST (i love this thing) :)

Any Info and fixes are greatly appreciated.

Greetings;
Torti

gyurco, 2024-01-11:
It works for me, using the WHDownLoad release. What core/game release are you referring to? But anyway, I'm not aware of any changes which could affect this.

I was not aware of the OSD bug, it can be surely fixed for the next time.

Torti-the-Smurf, 2024-01-11:
Hi Gyurco.
The core is not public yet (it has a few fixes that the public release doesn´t have).

Hmmmm; I have to investigate that further... Have to test a few more things to find the root...
Will be back later.

Greez & thanks

Torti-the-Smurf, 2024-01-11:
This is so weird. It has nothing to do with the MiniMig Core version. It always freezes at the same spot.
BTW, i did almost forget how slow old core is XD

If you like a faster and more fixed (by robinson), then go ahead ... 
https://www.mediafire.com/file/q7xrxjw30vb3gi0/Core.rbf/file

But back to Clown-O-Mania...
The weird thing is that the game worked before (whdload) on the same setup.

I did a test with a ADF and weirdly enough, that works...!!  
but the WHDload version freezes in the same setup. 
(all the same Cpu/chip speed setting as when testing the adf)

So odd.

Sadly i have no other SD card here to test but its a fine SanDisk.
I will reformat it... but i dont think that will make a difference.

Somthing went wrong with the Last Firmware on my Mist 1.4. , i guess.  
(i did the intermediate mid update, before upgrading to the most recent one)

It killed the clown ... well, kinda ;)

Anyway, greetings,
Torti

gyurco, 2024-01-11:
Try it with the intermediate firmware, it's the same as the latest before, just with a fix.
The only main thing the core depends on when running is the IDE/ATA, but it wasn't changed lately.

gyurco, 2024-01-11:
I tried Clown-O-Mania with this core, and works. Maybe yours is damaged on disk, try to re-extract it from the lzh.

Torti-the-Smurf, 2024-01-11:
I did the Intermedia FW update first and then, the 240105. 
(isnt that right ? .. i thought it has to be done this way...)

I also did test a fresh LHA into RAM: same thing happend... so it cant be the card... right ?
Did also test the same card in a WinUAE setting and there it works...

So odd...

Hmmm, so the thing that confuses me now is.. whats the the most recent FW ?
the intermediate or the 240105 ? (it is the newest from date alone... so it hast to be the last, right  ?)

a confused,  Torti :)

gyurco, 2024-01-11:
240105 is the latest
The intermediate is the previous.
But I hardly believe it causes the issue.
Try with a very old one to check it yourself (be aware if firmware change goes wrong, you have to use the USB to reflash). If you can identify the firmware version where it broke for you, then I can look at it. Otherwise, as I cannot reproduce, I have no idea.

gyurco, 2024-01-11:
The Wiki describes the unbricking process. Yes, an USB cable is enough.

Torti-the-Smurf, 2024-01-11:
I did a "update" with the intermediate ... (my heart was pounding....)

same thing happends.. the Clown wont show up. 
(freeze a few seconds into the STARBYTE logo, right before it should switch the the title screen))

I update once more to the 240105... (now i am always afraid to update the firmware) :)

The weird thing is.. I had a strange feeling when i did the 240105 update the first time .. 
.... like,; "should i really update, everything works just fine"...

and now, the clown refuses to play (via whdload ... even tried a older slave .. also JST)
well, i guess, "thats life" ;)

I did test around 200 games / demos with the recent firmware and this is the ONLY game that refuses now to work...  (i have still 600 games and a ton of demos left to test ... again...........)

so odd....

anyway, thanks for all the help and infos.
greetings, Torti

gyurco, 2024-01-11:
Starting from the intermediate, upgrades should be safe again, the bug is fixed.
But I still don't think it's about the firmware.
Try a clean boot (from a workbench disk), and check again.

Torti-the-Smurf, 2024-01-11:
I did already test that ;) also with no startup-sequence at all.
Same.

BUT.... now i have noticed something.
When i selected a file from this SD card to delete it, windows always showed...
Should this file with 0kb really be deleted...

THAT is not nowmal...
It always showed 0 when i tried to delete a file from this SD card. no matter how big the file was.

So, the next step was to use DISKPART.. and clean it.

NOW, the card dont know its size anymore and windows cant format it.

So i guess, the controller on the SD must have gone bad. (right?)

I buy a new one tomorrow and fingers crossed... it should work again... (hopefully)

Is this the first time a SanDisk let me down ?

Well, there is always a frist ... i guess ;)

Torti-the-Smurf, 2024-01-11:
Yupp, this card has definitely some problems.
i could "rescue" it with the SD Formater tool.
Now it shows the right size again... but even so.. when i try to delete a file, 
windows always tells me that the file i want to delete from this card is zero kb in size.

Well... that is something.... i guess.
I will report back tomorrow; with a brand new card. (that one was barely 1 year old)

Have a great day and HUGE thanks again for all the help and support.

greez,
Torti

gyurco, 2024-01-11:
One big deficiency of the firmware IDE that it ignores SD Card errors. So that must be very possible you came through a bad card.

Hope you'll get positive results, have a nice day.

Torti-the-Smurf, 2024-01-11:
I did more tests ... ( i could not resist) :)
Get this. It works with a FFS Filesystem partition on that same SD Card, but not with PFS3. 
(it used to work before the FW update, because i tested all the 833 games i have on it)

BTW, My windows 10 shows on all the exchangeable Flash USB devices a "Zero Size" 
when i want to delete a file. Weird, i never noticed that.

Anyway ; did you test Clown o mania WHD on a FFS Filesystem or PFS3 ?

Hmmm, what do you think....
Should i upload my Setup so you can check it out yourself. (your would love it, i am pretty sure ;)

So you can see for yourself. How about it ?
Cheers,
Torti

gyurco, 2024-01-11:
The firmware nothing has to do with the FS.
But if you can prove which version works and which not, then I'll investigate.

Torti-the-Smurf, 2024-01-11:
Pheeew. i did alot of testing... (like you said, the firmware relates to the IDE/ATA)

I make it short. please; try my setup. (just copy in on a SD Card and try clown o mania)

https://www.mediafire.com/file/hmckwdqmzjokcj9/MIST-The-Final-Cut.7z.001/file
https://www.mediafire.com/file/qe0gvf9hr1pou93/MIST-The-Final-Cut.7z.002/file
https://www.mediafire.com/file/68xbxi2qhll9iyv/MIST-The-Final-Cut.7z.003/file

When the Workbench has booted, simply press the right mouse button and choose the Game Menue.

Choose Clown o mania and see what happends .... man, i am curious .... (or is it just my mister?)

plz try.

thanks gyurco

gyurco, 2024-01-12:
Sorry, but it's huge. Can you do a smaller image with only this game?

Torti-the-Smurf, 2024-01-12:
Phew; after all this hours of testing i finally found the cause of the clown-o-mania Whdload freeze....
Its not the Firmware OR the SD Card.

Actually, its a Clown-o-Mania "DMA Wait code/slave bug" that can freeze the game in certain places.
It can happen in the StarBytes Logo.. or when entering a Cheat.. ect....

So, problem solved...
That was quiet an experience, i have to say XD

Anyway, HUGE thanks for the Support.
Greetings,
Torti