mist-devel/mist-binaries #149: Keyboard and mouse can hang in MiniMig

Torti-the-Smurf, 2024-01-30:
Hi.

A little question that bugs me since years :)    (minimig releated)

Is it the "fault" of the Core or the Firmware that the keyboard and mouse can get stuck/hang on an input.

Let me explain: With the mouse, i work on the Workbench and all the sudden, the last LMB get stuck 
(like its still pressed) and the only way to resolve it, is to move the mouse, then the Button will be released.

Same think happends with the keyboard, especially in Pinball games (not in pinball dreams)
but in pinball fantasies, Pinball Brain damage, pinball mania, obsession, ect...

just try to hit both flippers at the same time... and do that often ... you will see soon or later, 
depending on the game,
its either not possible to hit both keys at the exact same time... or that a input hangs 
(flipper is still up, even if nothing is pressed)

Any ideas ... ? and will it get fixed ; eventually ;) ?

Greetings and BIG thanks,
Torti

gyurco, 2024-01-30:
I never experienced such thing, also nobody reported it yet, so it must be a local thing to you.
The joystick stuck in OSD is already fixed in the firmware.

robinsonb5, 2024-01-30:
It's worth trying another keyboard and mouse.  (I have one mouse which behaves as you describe when used with Windows XP - but the same mouse is fine in Windows 10.  I'd have said it was impossible if I hadn't seen it myself.)

harbaum, 2024-01-31:
This is the behaviour I'd expect to see when USB problems happen. Then the packet indicating a key being released or a button isn't pressed, anymore may be lost.

The host then never learns that the key or button has been released until the next keyboard or mouse data packet being received. And this happens if you move the mouse or press another key.

Indeed, using a different device may solve the issue.

But this should then be independent of the running core or software.

Torti-the-Smurf, 2024-02-04:
Tested another USB Keyboard. Same result.
 
Those are the Pinball Games that have Issues : 
(pretty much the same problem in all of them ; cant press 2 keys at the same time, or hang very easily)
Pinball Mania
Obsession
Pinball Prelude
Brain Damage

and those work without any problem. !!
Pinball Dreams
Pinball Fantasies
Pinball Illusions
Slam Tilt.

No Keyboard problem anywhere, only in those  that are in the first List. Weids; some odd keyboard routines ?

Btw. I did set the Mouse "controls back to default"  (double cklick time)
and since then i had no more mouse hanging (or just luck).

greetings